adafruit/Adafruit_CircuitPython_LED_Animation #36: PixelMap, ranges of pixels - BUG:

cuppaCoffee1, 2020-06-06:
**Hardware:** Feather M4 Express ATSAMD51, 74AHCT125 Level-Shifter,168 NeoPixels [zig-zag configuration], 
**Software:** adafruit-circuitpython-feather_m4_express-en_US-5.3.0, adafruit-circuitpython-bundle-5.x-mpy-20200527

**Animation types that work with PixelMap range of pixels:**
Solid, Blink, ColorCycle, Pulse,  

**Animation types that display as a single color at a time - for the whole PixelMap range of pixels:**
Rainbow, Chase, RainbowChase, RainbowSparkle

**Animation types that error out with PixelMap range of pixels:**
Comet, RainbowComet (ValueError: Slice and input sequence size do not match.)
Sparkle, SparklePulse (ValueError: Sparkle needs at least 2 pixels)

-------------------------------------------------------------------
**Test Code:**
import board
import neopixel
from adafruit_led_animation.sequence import AnimationSequence
from adafruit_led_animation.helper import PixelMap
from adafruit_led_animation.animation.solid import Solid
from adafruit_led_animation.animation.colorcycle import ColorCycle
from adafruit_led_animation.animation.blink import Blink
from adafruit_led_animation.animation.comet import Comet
from adafruit_led_animation.animation.chase import Chase
from adafruit_led_animation.animation.pulse import Pulse
from adafruit_led_animation.animation.rainbow import Rainbow
from adafruit_led_animation.animation.rainbowchase import RainbowChase
from adafruit_led_animation.animation.rainbowcomet import RainbowComet
from adafruit_led_animation.animation.rainbowsparkle import RainbowSparkle
from adafruit_led_animation.animation.sparkle import Sparkle
from adafruit_led_animation.animation.sparklepulse import SparklePulse
from adafruit_led_animation.color import PURPLE, AMBER, TEAL, PINK, MAGENTA

pixels = neopixel.NeoPixel(board.D6, 168, brightness = .1, auto_write=False)

section_a = PixelMap(pixels, [(0, 56)])
section_b = PixelMap(pixels, [(56, 112)])
section_c = PixelMap(pixels, [(112, 168)])

solid = Solid(section_a, color=PINK)
blink = Blink(section_b, speed=0.5, color=TEAL)
colorCycle = ColorCycle(section_c, speed=0.4, colors=[MAGENTA, PURPLE, TEAL])
pulse = Pulse(section_a, speed=0.1, color=TEAL, period=3)
rainbow = Rainbow(section_b, speed=0.1, period=1)
chase = Chase(section_c, speed=0.1, color=PURPLE, size=3, spacing=6)
rainbowChase = RainbowChase(section_a, speed=0.1, size=5, spacing=3)
rainbowSparkle = RainbowSparkle(section_b, speed=0.1, num_sparkles=15)
# comet = Comet(section_c, speed=0.01, color=PURPLE, tail_length=10, bounce=True)
# rainbowComet = RainbowComet(section_a, speed=0.01, tail_length=7, bounce=True)
# sparkle = Sparkle(section_b, speed=0.05, color=AMBER, num_sparkles=10)
# sparklePulse = SparklePulse(section_c, speed=0.001, period=3, color=AMBER)

animations = AnimationSequence(solid, blink, colorCycle, pulse, rainbow, chase, rainbowChase, rainbowSparkle, 
# comet, rainbowComet, sparkle, sparklePulse,
advance_interval=3, auto_clear=True)

while True: 
    animations.animate()

cuppaCoffee1, 2020-06-06:
Spare thought, for _PixelMap ranges of pixels_ . Maybe an alternative would be to modify the Helpers so that you could describe a range.

The horizontal and vertical helpers have a starting boundary at pixel 0. They can be shortened on their far side - by describing their width or height with less than the total number of x or y pixels in the matrix. The starting boundary is always at pixel 0.  If the Helper's origin could be offset in x or y, this could possibly be a different way of creating an animation ranges of pixels.

rhooper, 2020-06-26:
For pixel ranges, does PixelSubset solve your problem?
https://github.com/adafruit/Adafruit_CircuitPython_LED_Animation/blob/master/adafruit_led_animation/helper.py#L306

The other issues may be because the effective pixel strip length of the PixelMap was 1 pixel.  I'll have a look through the animations and add a good error message to animations that need more than 1 pixel.

cuppaCoffee1, 2020-06-27:
Yes it does! I tried out all of the animation types that I said were showing as one color, or error-ing out and they all work with PixelSubset. I don't remember why I started using PixelMap. I think I saw some sample code in the reference guide and thought that was the way to go. Thanks rhooper!

My test code:
 `
import board
import neopixel
from adafruit_led_animation.helper import PixelSubset
from adafruit_led_animation.animation.comet import Comet
from adafruit_led_animation.animation.rainbowchase import RainbowChase
from adafruit_led_animation.animation.sparkle import Sparkle
from adafruit_led_animation.color import BLUE, RED
from adafruit_led_animation.group import AnimationGroup

pixels = neopixel.NeoPixel(board.D6, 168, brightness=.2, auto_write=False)

block1 = PixelSubset(pixels, 0, 56)
block2 = PixelSubset(pixels, 56, 112)
block3 = PixelSubset(pixels, 112, 168)

comet1 = Comet(block1, speed=.001, color = RED, tail_length=12, bounce=False)
rainbow = RainbowChase(block2, speed=0.01, size=56, spacing=12)
sparkle = Sparkle(block3, speed=.01, color = BLUE)

animations = AnimationGroup(
    AnimationGroup(
        comet1,
        sparkle,
        rainbow,
    )
)

while True:
    animations.animate()

`
[image]

closed 2020-06-30 by dherrada