adafruit/Adafruit_CircuitPython_LED_Animation #41: PixelSubset numbering range is too large by one

jedgarpark, 2020-06-09:
PixelSubset numbering currently requires a value one higher than the last pixel you want in a slice. This is how it currently works:
```
pixel_ringA = PixelSubset(pixels, 0, 24)  # 24 pixel ring
pixel_stripA = PixelSubset(pixels, 24, 40)  # 16 pixel strip
pixel_stripB = PixelSubset(pixels, 40, 56)  # 16 pixel strip
```
I would expect it to work like this:
```
pixel_ringA = PixelSubset(pixels, 0, 23)  # 24 pixel ring
pixel_stripA = PixelSubset(pixels, 24, 39)  # 16 pixel strip
pixel_stripB = PixelSubset(pixels, 40, 55)  # 16 pixel strip
```

cuppaCoffee1, 2020-06-09:
@jedgarpark re:pixel slices, ranges

Warning, novice chiming in here. I have worked with slices, ranges, as well as PixelSubset on my neopixel matrix and all these seem to require overlapping numbers to work [see photo].  The matrix is a single strand of 168, which divides nicely into 3 x 56. Example 2 shows how you would intuitively think the numbering would work, but it offsets back 1 pixel [note: observing one range at a time, they appear as in photo, there is no overlapping of pixels]

Example 1
-------------------
import board
import neopixel
import random

numpix = 168
pixels = neopixel.NeoPixel(board.D6, numpix, auto_write=False)

def range1():
    return random.randrange(0, 56, 1)

while True:
    pixels[range1()] = (0, 4, 4)
    pixels[range1()] = (0, 8, 8)

    pixels[112:168:1] = [(0, 0, 2)] * (56)
    pixels[56:112:1] = [(10, 10, 0)] * (56)
    pixels.show()

Example 2
-------------------
numpix = 168
pixels = neopixel.NeoPixel(board.D6, numpix, auto_write=False)

def range1():
    return random.randrange(0, 55, 1)

while True:
    pixels[range1()] = (0, 4, 4)
    pixels[range1()] = (0, 8, 8)

    pixels[111:167:1] = [(0, 0, 2)] * (56)
    pixels[55:111:1] = [(10, 10, 0)] * (56)
    pixels.show()

[image]

tannewt, 2020-06-09:
@jedgarpark This is common to Python. Ranges are start inclusive and end exclusive.

jedgarpark, 2020-06-11:
OK, then we'll just need to be clear in documentation since this "feels" different than the way we normally learn to think about NeoPixel ranges.

jedgarpark, 2020-06-11:
Thanks @cuppaCoffee1 and cool looking matrices!

closed 2020-06-11 by jedgarpark

cuppaCoffee1, 2020-06-11:
Thanks @jedgarpark! I'm thinking the difference might be the way ranges are defined, vs. a pixel index.