adafruit/Adafruit_CircuitPython_LED_Animation #42: Special support for "ring" that could benefice some effect.

dglaude, 2020-06-09:
I noticed that some effect do not look "perfect" when your row of RGB LED make a loop or is circular, like this product: https://www.adafruit.com/product/1586

When you get to the last pixel, the next pixel is the first one.
With current library, if you run an effect multiple time, the effect will end on last pixel and a new effect will start on the first pixel. This break the "motion". For Meteor, this mean a new meteor will be launched, rather than to have one meteor that start and rotate again and again.

I believe some effects might benefit from beeing continuous from one run to the other.

rhooper, 2020-06-26:
@dglaude Good idea.  I'll grab one of my rings and see what I can do.  Were there specific animations other than Comet you wanted to see ring support on?

dglaude, 2020-06-27:
Comet was the most obvious.

Chase can be made OK by choosing the right size and spacing.

I believe the Rainbow is OK since last color in the wheel is near the first one, so you get a rotating rainbow.

But I guess RainbowComet fail like Comet.

I think you have good visual example of circular usage on the following page from the guide: https://learn.adafruit.com/circuitpython-led-animations/animation-group

closed 2020-08-15 by rhooper