adafruit/Adafruit_CircuitPython_LED_Animation #43: Gradient Animation Request

firepixie, 2020-06-12:
I'd love a gradient animation. I want to be able to select my colors (maybe 3-4, or maybe more for a whole palette). The colors should fade into each other seamlessly, and I want to be able to choose the period / number of LEDs per section. Then I'd want the whole thing to scroll along the strip and be able to choose the speed.

Gradient (pixels, speed=0.1, colors=(color.RED, color.BLUE, (0,200,10), 0XFF0000), period=5)

rhooper, 2020-06-26:
@firepixie Do you have enough memory on the board(s) you want to use to import FancyLED?  https://github.com/adafruit/Adafruit_CircuitPython_FancyLED

If yes, then I think this should be pretty easy to implement.

firepixie, 2020-07-01:
Yes I believe so. Incorporating FancyLED would be perfect!

firepixie, 2020-08-19:
Following up on this - any way to get some sample code on how to use FancyLED to make gradients? :)

dglaude, 2020-08-19:
In this thermal camera project, I use FancyLED to make gradients. This is a link to a version that does not require the thermal camera, so you could be able to test it: https://github.com/dglaude/CircuitPython_MLX90640_PyGamer_Plus/blob/master/fake_mlx90640_pygamer%2Bsd.py

In the published version in the learn guide, I do something similar without FancyLED (the credit/link for the code sample I used): https://learn.adafruit.com/adafruit-mlx90640-ir-thermal-camera/circuitpython-thermal-camera

I hope it help.