adafruit/Adafruit_CircuitPython_LED_Animation #53: adafruit_led_animation

rdagger, 2020-06-22:
After upgrading to the latest Circuit Python Bundle (circuitpython-bundle-5.x-mpy-20200621), I'm getting the following error with the **adafruit_led_animation** library on the Feather nRF52840 Express:
_File "code.py", line 7, in <module> TypeError: 'module' object is not callable_
Here is a sample program to reproduce the error:
```
from adafruit_led_animation.animation import Pulse, Solid
import adafruit_led_animation.color as color
from neopixel import NeoPixel
from time import sleep
from board import NEOPIXEL
pixel = NeoPixel(NEOPIXEL, 1)
solid = Solid(pixel, color.GREEN)  # THIS LINE THROWS THE ERROR
solid.animate()  
sleep(4)
```
The code works with circuitpython-bundle-5.x-mpy-20200516
The code stops working with circuitpython-bundle-5.x-mpy-20200518

I'm running CircuitPython 5.4.0-beta.1-43-g1504d9005 on 2020-06-22; on Feather nRF52840 Express.

tannewt, 2020-06-22:
Looks like it was 2.0.0 that was included. The notes are here: https://github.com/adafruit/Adafruit_CircuitPython_LED_Animation/releases/2.0.0

Here is an example change: https://github.com/adafruit/Adafruit_CircuitPython_LED_Animation/commit/cbfa114e7d104a6757d3ab749153d1b284a9c8c2#diff-879798843e31cea6b37d583453edd26f

It looks like you may need to modify the first import for another level of abstraction.

rdagger, 2020-06-23:
Thanks that worked!  Unfortunately, it looks like the _min_intensity_ and _max_intensity_ parameters have been removed from the **Pulse** animation.  Any chance of getting those back?  It really helps to dial in a smooth LED pulse.

```
class Pulse(Animation):
    """
    Pulse all pixels a single color.
    :param pixel_object: The initialised LED object.
    :param int speed: Animation refresh rate in seconds, e.g. ``0.1``.
    :param color: Animation color in ``(r, g, b)`` tuple, or ``0x000000`` hex format.
    :param period: Period to pulse the LEDs over.  Default 5.
    :param max_intensity: The maximum intensity to pulse, between 0 and 1.0.  Default 1.
    :param min_intensity: The minimum intensity to pulse, between 0 and 1.0.  Default 0.
    """

class Pulse(Animation):
    """
    Pulse all pixels a single color.
    :param pixel_object: The initialised LED object.
    :param float speed: Animation refresh rate in seconds, e.g. ``0.1``.
    :param color: Animation color in ``(r, g, b)`` tuple, or ``0x000000`` hex format.
    :param period: Period to pulse the LEDs over.  Default 5.
    """
```

rdagger, 2020-06-23:
Also the _min_intensity_ and _max_intensity_ parameters are still present in the **SparklePulse** animation but don't appear to be wired up.

rhooper, 2020-06-26:
Good catch on SparklePulse.  

Restoring min/max intensity shouldn't be difficult.  

With regards to the import issue, I'd love to add backward compatibility for imports, but will need to test see how that affects memory use.

FoamyGuy, 2024-12-20:
Resolved #116

closed 2024-12-20 by FoamyGuy