adafruit/Adafruit_CircuitPython_LED_Animation #56: cycle_complete is not working for Chase animation

cjsieh, 2020-06-22:
Recently changes were made in how  anim.draw_count is updated in animation/__init__.py  .  Near line 102/105 anim.draw_count  is incremented.  Thus it is incremented 2 times.  
The issue is that in chase.py at line 119 a check is done on draw_count .  This check does a % with "num of pixels" .  This test is never true now that draw_count is incremented  twice and thus cycle_complete is never set.
Solutions are to remove the anim.draw_count increment at line 105 in animation/__init__.py  or modify chase.py to handle this change of draw_count .  
I can provide PR for either way,  which would you prefer or  something else.  Both comet and sparkle work with the suggested change to animation/__init__.py .

dherrada, 2020-06-25:
I'll try to take a look at this tomorrow

cjsieh, 2020-06-26:
To reproduce this issue modify  example/led_animation_sequence.py  
replace in line 29
"advance_interval=3"
with
"advance_on_cycle_complete=True"

Results are blink and comet run 1 time each 
chase then keeps running because cycle_complete is never triggered

As mentioned one solution is to comment out line 105 in animation/__init__.py

rhooper, 2020-06-26:
Good catch, @cjsieh.  If you want to submit a PR, we can get your fix merged!

cjsieh, 2020-06-28:
closed via PR #58

closed 2020-06-28 by cjsieh