adafruit/Adafruit_CircuitPython_LED_Animation #65: Updating brightness mid-animation

stickperson, 2020-09-11:
Any idea how difficult it would be to make it easier to update the brightness of the animation once it has been created? I'm using a `RainbowComet` animation with a `PixelSubset` . While that's running I also have a customized `Pulse` animation running on another subset of pixels. As `pulse.draw()` is called I keep track of the current brightness level so that my other LEDs can change as well.

I'd like the brightness of the `RainbowComet` animation to be in sync with that of my customized `Pulse` animation. To do that, I created a slightly customized `RainbowComet` class that overrides the `draw` method to use the current brightness:

```
    def draw(self):
        """
        Same exact implementation as the base class except adding in current brightness
        """
        # super().draw()
        colors = self._comet_colors
        if self.reverse:
            colors = reversed(colors)
        for pixel_no, color in enumerate(colors):
            # The next three lines were added to add in brightness
            if len(color) == 3:
                color = list(color)
                color.append(self.displayer.current_brightness / 100)

            draw_at = self._tail_start + pixel_no
            if draw_at < 0 or draw_at >= self._num_pixels:
                if not self._ring:
                    continue
                draw_at = draw_at % self._num_pixels

            self.pixel_object[draw_at] = color

        self._tail_start += self._direction

        if self._tail_start < self._left_side or self._tail_start >= self._right_side:
            if self.bounce:
                self.reverse = not self.reverse
                self._direction = -self._direction
            elif self._ring:
                self._tail_start = self._tail_start % self._num_pixels
            else:
                self.reset()
            if self.reverse == self._initial_reverse and self.draw_count > 0:
                self.cycle_complete = True
```

I tried `animation._pixel_object.brightness` but that ended poorly.

steveof2620, 2023-06-08:
Would also like a way to adjust the speed within the while True loop.

FoamyGuy, 2024-12-21:
I took a look into this, having a `brightness` property would be quite tricky because the "current brightness level" state is only stored inside of the `pulse_generator()` the `Pulse` class itself doesn't have access to that. And the brightness in there isn't just a value 0 - 1.0 like `pixels.brightness` instead instead it's 0-950 integers, you could map it to the range 0.0 - 1.0 if you wanted though. 

The closest thing that you can do today, and the way I would probably recomend to approach the behavior noted is by accessing the color of the PixelSubset that is running the Pulse animation. 

`pulse_subset[0]` will give you the current color of the pixels in the subset that is running the Pulse. From that color you'd have to convert it "backwards" to find out what brightness level it was for in order to use that information on the other subset. 

I'm going to close because I think it would be quite tricky to implement, and ultimately will give you the basic information as accessing the color of the pixels could do today without any changes. 

With regards to @steveof2620's comment: I have submitted #123 to add `period` property to Pulse and SparklePulse which were the 2 remaining animations that had a period argument, but not property yet. This new property allows you to update the speed of the full pulse cycle for those animations.

closed 2024-12-21 by FoamyGuy