adafruit/Adafruit_CircuitPython_LED_Animation #71: The colors provided by adafruit_led_animation.color are not documented

siddacious, 2020-11-27:
Several colors are used in examples, however the full list is not available anywhere other than in the code.

It would be great if these could be documented so users know what is available to use.

tannewt, 2020-11-30:
I think they'd appear in ReadTheDocs if they had `"""    """` comments afterwards.

johnwargo, 2020-12-01:
They're all listed in this file: https://github.com/adafruit/Adafruit_CircuitPython_LED_Animation/blob/master/adafruit_led_animation/color.py

Sure, the list could be in the docs, but this is open source, so its really easy to find the code that defines the colors.

siddacious, 2020-12-02:
@johnwargo Yes, one could just read the code but we're trying to make this accessible to people who are still learning and for whom reading code doesn't come as easily as those with more experience.

kattni, 2020-12-08:
I thought I had documented them in the guide - I did not. That needs to be rectified regardless of updating the API documentation.

That said, I will look into getting them to show up in the API documentation as well. I noticed this two weeks ago when working on a guide where the code allowed for updating colors, and I realised they weren't listed in the docs.

PR #73 (2020-12-08): Renders color.py in Sphinx documentation.
Fixes #71
closed 2020-12-08 by ladyada