adafruit/Adafruit_CircuitPython_LED_Animation #74: Reverse setting in Comet not working

chrisw63, 2020-12-21:
Pretty new at this - please be gentle if I have this wrong.  When I run the Comet animation (using MagTag from ADABOX017), and set the Reverse option, the LED's seem to stay dark:
<Comet(strip, strip_comet_speed, comet_one_color, tail_length=strip_comet_tail, reverse=True)>

The draw() function is where the action is:
<def draw(self):
        colors = self._comet_colors
        if self.reverse:
            colors = reversed(colors)
        for pixel_no, color in enumerate(colors):
            draw_at = self._tail_start + pixel_no
            if draw_at < 0 or draw_at >= self._num_pixels:
                if not self._ring:
                    continue
                draw_at = draw_at % self._num_pixels

            self.pixel_object[draw_at] = color

        self._tail_start += self._direction

        if self._tail_start < self._left_side or self._tail_start >= self._right_side:
            if self.bounce:
                self.reverse = not self.reverse
                self._direction = -self._direction
            elif self._ring:
                self._tail_start = self._tail_start % self._num_pixels
            else:
                self.reset()
            if self.reverse == self._initial_reverse and self.draw_count > 0:
                self.cycle_complete = True>

_Bounce_ works using _reverse_.  _Reverse_ (alone) does not.  Debug print statements show normal numbers for Comet with no options.  With _reverse_ set, __tail_start_ is being set to 45 (30 LED's plus 15 pixel 'tail'), decremented to 44 by <self._tail_start += self._direction> and reset to 41 elsewhere before _draw()_ is called again. It appears the last IF statement is the cuplrit.  With no support for simple 'reverse' in the root IF block (last ten lines), the last IF resets the animation.  If reverse is simply there to be a flag for bounce, it should be removed from the public settable parameters. If it's meant to function on it's own, this root IF block needs fixing to support it.

plugowski, 2021-05-02:
Same problem here. Reverse with Comet as well as with RainbowComet doesn't work.
I tried to set reverse during initialize and also change it on initialized object:

```
animation = RainbowComet(pixels, speed=0.1, tail_length=10, reverse=False)
animation.reverse = True
```

plugowski, 2021-05-03:
@chrisw63 I've already fixed this issue for myself, now I'm waiting for merge of my PR: https://github.com/adafruit/Adafruit_CircuitPython_LED_Animation/pull/78

kattni, 2021-11-03:
This has been resolved.

closed 2021-11-03 by kattni