adafruit/Adafruit_CircuitPython_LED_Animation #111: consider making keyword-only parameters for some animations

tlyu, 2023-01-25:
Recent issues like #108, which seem to be the result of API changes in a base class, might be mitigated if the less-used parameters for animation classes like `Comet` were declared as keyword-only.

tlyu, 2023-01-25:
This is admittedly a breaking API change for anyone depending on supplying positional arguments for parameters that have defaults. Most (all?) of the examples seem to use keyword arguments for everything except the pixels object. Arguably, the API has already recently been broken in this way with the addition of the `background_color` parameter to `Comet`.

tekktrik, 2023-01-27:
I think this is a good change, but needs to be weighed against what would change in terms of existing Learn guides and other documentation.  @kattni, tagging you about this as a heads up, happy to talk about what that would entail.