adafruit/Adafruit_CircuitPython_LED_Animation #112: sparkle.mpy animation broken / just flickering

GithubUser99999999, 2023-02-21:
`sparkle = Sparkle(pixels, speed=0.05, color=AMBER, num_sparkles=10)`

This code should look like this (code and video taken from the adafruit guide):

https://user-images.githubusercontent.com/101017273/220235696-7bfdf958-84ca-4982-ad35-4bb50bd12839.mp4

But it does actually look like this:

https://user-images.githubusercontent.com/101017273/220235797-7f151fe9-4d0e-416b-8df5-ded4ece417f8.mp4

However, the rainbowsparkle.mpy animation does indeed work (with the same code):

https://user-images.githubusercontent.com/101017273/220235979-aa422634-d5fa-4db8-9359-5fd1cf343ed2.mp4

Tested with CircuitPython 8 on a PICO RP2040

Is this animation currently broken?

GithubUser99999999, 2023-03-01:
Tested on a Gemma M0 with the same result:

https://user-images.githubusercontent.com/101017273/222038523-bbecdf56-fb6a-456b-a51c-da63d532403c.mp4

It only has enough memory to run the standard Sparkle animation, but not SparklePulse or RainbowSparkle to show the difference.

(I tried an ItsyBitsy RP2040 too, same result and Sparkle doesn't work, but SparklePulse and RainbowSparkle do work.)

GithubUser99999999, 2024-08-14:
almost 1,5 years later with CircuitPython 9.1.1 and the latest libraries this bug is still not fixed

GithubUser99999999, 2024-08-14:
SparklePulse seems to just call Sparkle and Pulse and for some reason it works.

https://github.com/adafruit/Adafruit_CircuitPython_LED_Animation/blob/main/adafruit_led_animation/animation/sparklepulse.py

If I could disable the pulsing, it would work.

https://github.com/adafruit/Adafruit_CircuitPython_LED_Animation/blob/main/adafruit_led_animation/animation/sparkle.py

Or there is some bug I don't recognise in here.

dhalbert, 2024-08-14:
@GithubUser99999999 Just checking, are you the same person as https://forums.adafruit.com/viewtopic.php?p=1025640#p1025640 ?

dhalbert, 2024-08-15:
After testing Sparkle, Pulse, and SparklePulse, I think the problem here is that the videos in the guide on that page are mislabeled. The [Sparkle effect at the very top of the page](https://learn.adafruit.com/assets/91484) is what you should expect sparkle to look like: very quick random transitions. The animation labeled as

dhalbert, 2024-08-15:
After testing Sparkle, Pulse, and SparklePulse, I think the problem here is that the videos in the guide on that page are mislabeled. The [Sparkle effect at the very top of the page](https://learn.adafruit.com/assets/91484) is what you should expect sparkle to look like: very quick random transitions. The animation [labeled as Sparkle](https://learn.adafruit.com/assets/91421) is really SparklePulse. And the animation [labeled as SparklePulse](https://learn.adafruit.com/assets/91422) is something else, maybe just an adjusted SparklePulse.

GithubUser99999999, 2024-08-15:
Aaaaaah okay. I never would've figured that out ^^ Thanks for that.

Do you perchance have an idea of how to use the RainbowSparkle library/code with just one single colour then?

The sparkling of SparklePulse and RainbowSparkle is perfect, but not that of Sparkle, that's just flickering.

dhalbert, 2024-08-15:
> I don't seem to be able to create any animation like the one in the gif from the guide at all. The guide gif also is simply not pulsing, it's just sparkling with one single colour.

Just to be clear, which one do you mean?

GithubUser99999999, 2024-08-15:
https://cdn-learn.adafruit.com/assets/assets/000/091/421/original/leds_LED_Anim_Sparkle.gif?1590196528

A non-pulsing single-colour sparkle animation like this. RainbowSparkle works fine, if only I knew how to change the code so that it outputs just one colour.

https://github.com/adafruit/Adafruit_CircuitPython_LED_Animation/blob/main/adafruit_led_animation/animation/rainbowsparkle.py

dhalbert, 2024-08-15:
I tried making a new video of the Sparkle effect and did not succeed in a faithful representation. I think the video in the guide may indeed be Sparkle, but because of video aliasing effects, it does not show up well and looks slower than in real life. The video at the top of that page is the best representation of Sparkle. I've added a warning to the guide about that video.

I also tried some simple modifications to the example code to force a monochromatic `RainbowSparkle`, but it wasn't possible because of the way the code is structured. The "rainbow" feature made it not possible to specify the individual color. If you change the initialization to
```py
sparkle_pulse = SparklePulse(pixels, speed=0.05, period=3, color=JADE, step=256)
```
you will get a monochromatic red `SparklePulse`, but changing the initial color away from red requires modifications to both `Rainbow` and `RainbowSparkle`, or really, a thorough restructuring. I encourage you to study the current animation classes and experiment with modifying them.

GithubUser99999999, 2024-08-15:
I am using this as a sparkly replacement now even though it's not as random. 

`sparkle = Chase(pixel_spiral[0], speed=0.2, color=FARBE, size=4, spacing=8)`

https://github.com/user-attachments/assets/07e8f100-0785-49fb-8026-8e1cfd47c0bf

I would need white for this for example with yellow PLA. Thanks for your help anyway!

Maybe I will be good enough to understand how the rainbowsparkle code actually works in-depth in order to be able to modify it for single colour use.

dhalbert, 2024-08-15:
I am going to close this for now as it is not a bug in the library, but was more of a Learn Guide documentation problem. Feel free to continue posting.

DJDevon3, 2024-08-16:
Please take a look at my [proposed fix](https://github.com/DJDevon3/My_Circuit_Python_Projects/tree/main/Boards/raspberrypi/Adafruit%20Feather%20RP2040%20Prop-Maker/SingleSparkleFix).  Only file edited was sparkle.py besides obvious changes in code.py because I'm using a prop-maker rp2040 feather.

GithubUser99999999, 2024-08-17:
It's going in the right direction but not quite. Here are some videos:

original Adafruit Sparkle animation (speed=0.05, color=WHITE, num_sparkles=1, brightness=1)
https://github.com/user-attachments/assets/6ad0a996-ad5b-44b0-b896-eb8eba48bb39
-> doesn't sparkle at all, just flickering as if broken

your edited Sparkle animation (speed=0.05, color=WHITE, num_sparkles=1, brightness=1)
https://github.com/user-attachments/assets/6fa3c5ab-4d5f-43aa-8a3c-e14357c9ba92
-> it does actually sparkle, but it's still very fast and also very dark

here as an example the Adafruit RainbowSparkle animation with Step=255 to show only RED (speed=0.05, num_sparkles=10, brightness=1)
https://github.com/user-attachments/assets/ea0b7506-4f16-4149-84f7-b0891b021b89
-> very bright single colour sparkling

DJDevon3, 2024-08-17:
Seems like the only thing you want is a slower sparkle and keeping the dim color so the star is illuminated.  If that's the case then just change speed from 0.05 to something like 0.1 or 1.0  Speed is time in seconds so 0.05 is very fast and 1.0 is 1 second.