adafruit/Adafruit_CircuitPython_LED_Animation #113: Changing rainbow period on the fly

srmo, 2023-05-26:
Ahoi! I'm just starting out with circuitpython and have to say thank you for those amazing libraries. 
Easy to use and setup and looks great.

Now I'm hitting a wall. I want to change the period of a Rainbow on the fly. The property _period isn't the way to go.
`speed` could be used but this isn't what I want as I want to speed up the color transitions.

I've found a way to use the cycle complete receiver callback logic I.e. I create a new rainbow with the new period on a complete cycle.
But this leads to quicker period changes for quicker cycles...erm.

OK, what is my goal here: using the feather nRF52840 express, I create a webserver on a raspi4 which conencts via bluetooth to the feather and provides a WebUI to control brightness and speed of my WS2811 running the rainbow animation.
The webserver will receive POST requests and translate them into some UART packets.

This works fine for brightness as I can directly access `pixels.brightness` mid-animation but I see no way to change the period.

What am I missing? How should I go about it?

FoamyGuy, 2024-12-13:
@srmo Hello, sorry it's been a while before any response. Setting `rainbow._period` to a new value was the way to achieve this, but you would also need to call `rainbow.reset()` after updating `_period` in order for the new value to actually have an effect. 

It's not ideal to access and change "private" fields with a leading underscore in their name though, and there doesn't seem to be a specific reason not to allow the user to update the period configuration anyhow. 

I've submitted #121 that makes it into a public property and adds the reset call to the setter so that user code won't have to do that "manually"

closed 2024-12-13 by FoamyGuy

srmo, 2025-01-24:
Hey @FoamyGuy ! Thanks for coming back to this and no worries about the delay, I know how it is. I'll will give this a whirl once I've re-established my setup (been moving places and currently putting all of it back together again :) )

PR #121 (2024-12-13): public property for rainbow.period
@ladyada 
Resolves: #113 

`period` was providing as an arg to init and then stored in private field. This allows it to be accessed and updated more easily, with the requisite call to `reset()` to make it have an effect after being updated.