adafruit/Adafruit_CircuitPython_LED_Animation #117: Pulse: ValueError: Incompatible .mpy file.

samzmann, 2024-02-03:
I'm trying to import the Pulse animation module, but getting error. Other animation modules work.

I'm using circuitpython 8.2.9 on a raspi pico, and on v2.8.0 of the animation lib.

I'm using the CircuitPython VSCode extension, this is my config:
```json
{
    "python.languageServer": "Pylance",
    "python.analysis.diagnosticSeverityOverrides": {
        "reportMissingModuleSource": "none",
        "reportShadowedImports": "none"
    },
    "python.analysis.extraPaths": [
        "/Users/sam/.vscode/extensions/joedevivo.vscode-circuitpython-0.2.0-darwin-arm64/boards/0x239A/0x80F4",
        "/Users/sam/.vscode/extensions/joedevivo.vscode-circuitpython-0.2.0-darwin-arm64/stubs",
        "/Users/sam/Library/Application Support/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20240130/adafruit-circuitpython-bundle-py-20240130/lib"
    ],
    "circuitpython.board.version": "8.2.9",
    "circuitpython.board.vid": "0x239A",
    "circuitpython.board.pid": "0x80F4"
}
```

When I try to import `pulse.py` manually, I get an error at this line:
https://github.com/adafruit/Adafruit_CircuitPython_LED_Animation/blob/251bcd13cf14da68bd176d6d2d97b844fac5cdd5/adafruit_led_animation/animation/pulse.py#L78

I tried fiddling with it, doing top level import, removing it, but every time something else breaks...

dhalbert, 2024-03-26:
Please try installing the 8.x version of the library from the 8.x bundle or via `circup`. I don't know what the VSCode extension knows about library versions. It looks like it's referencing the .py version of the bundle, but you might have the 9.x (or some very old) version of the animation library on the board.

The 0.2.0 of the VSCode extension is problematic and people have had to roll back to 0.1.2. But the problems had to do with the serial terminal.

dhalbert, 2024-07-06:
Presuming resolved, so closing.

closed 2024-07-06 by dhalbert